sandeepmistry/node-chip-io #1: Is PWM implemented yet?

deadprogram, 2016-02-05:
I noticed that this function https://github.com/sandeepmistry/node-chip-io/blob/26b356b7a6c1054302f58eb33df60acbffa82ab7/lib/chip-io.js#L105 does not appear to have an associated implementation in either https://github.com/sandeepmistry/node-chip-io/blob/master/lib/axp209.js or https://github.com/sandeepmistry/node-chip-io/blob/master/lib/pcf8574a.js

Am I missing something here? Thanks for the help.

sandeepmistry, 2016-02-07:
@deadprogram nope, not implemented.

Please see the read me for what is currently supported: https://github.com/sandeepmistry/node-chip-io#pin-guide

closed 2016-02-07 by sandeepmistry

deadprogram, 2016-02-07:
That being the case, shouldn't you remove the code at https://github.com/sandeepmistry/node-chip-io/blob/26b356b7a6c1054302f58eb33df60acbffa82ab7/lib/chip-io.js#L105-L117

sandeepmistry, 2016-02-07:
@deadprogram why? What if the feature is coming in the future?

deadprogram, 2016-02-07:
At first, since the main object has that interface, I thought it was available. Only after trying it, and not succeeding, did I learn by looking thru the code that is was not yet implemented. Anyhow, just a suggestion.