sandeepmistry/node-chip-io #2: Possibly simplified temperature example

rwaldron, 2016-02-14:
I don't have a CHIP here to test this on, but I believe that example could be simplified to: 

```js
var five = require('johnny-five');
var Chip = require('chip-io');

var board = new five.Board({
  io: new Chip()
});

board.on('ready', function() {
  var temp = new five.Sensor("INTTEMP");
  var c = 0;

  temp.on("change", function() {
    c = this.value * 0.1 - 144.7;
    console.log('Internal temperature is ' + c.toFixed(2) + ' °C');    
  });
});
```

If someone could try that out for me, that would be awesome!

sandeepmistry, 2016-02-16:
@rwaldron thanks for the suggestion! It makes the example much simpler.

I've pushed the change for the battery voltage example (https://github.com/sandeepmistry/node-chip-io/commit/03b645829ea75c04ec649e36cab3c1cf0869acc0).

However, I need to take another look at the internal temperature code, without your suggestion after a few seconds it flip-flops between 56.70 °C and -137.30 °C. (Nice to see ```five.Sensor``` collects samples and uses the median).

rwaldron, 2016-02-16:
Another option would be to have Chip-IO provide its own controller for `Thermometer` class. I've been wanting to experiment doing this for other io plugins, eg. Particle Core/Photon could provide a controller for its RGB LED, that could be used with `Led.RGB`); until now I just haven't been very motivated to write the code. I will open an issue for that and submit a patch with implementation and examples for you to review

sandeepmistry, 2016-02-19:
Closed via #3.

closed 2016-02-19 by sandeepmistry