sandeepmistry/node-chip-io #5: Unable to install chipo-io

RyanOC, 2016-05-30:
Would you be able to assist in an issue I'm having installing chip-io? I've completed all of the instructions you have but still get errors. Here is the output...

chip@chip:~/Documents/sandbox/nodetest01$ npm install chip-io
npm WARN package.json nodetest01@0.0.0 No description
npm WARN package.json nodetest01@0.0.0 No repository field.
npm WARN package.json nodetest01@0.0.0 No README data
\
> ref@1.3.2 install /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ref
> node-gyp rebuild

/bin/sh: node: command not found
gyp: Call to 'node -e "require('nan')"' returned exit status 127. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:344:16)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.3.0
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ref
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! Error: ENOENT, utime '/home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ffi/deps/libffi/testsuite/libffi.call/cls_4byte.c'
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.3.0
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "chip-io"
npm ERR! cwd /home/chip/Documents/sandbox/nodetest01
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ffi/deps/libffi/testsuite/libffi.call/cls_4byte.c
npm ERR! fstream_path /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ffi/deps/libffi/testsuite/libffi.call/cls_4byte.c
npm ERR! fstream_type File
npm ERR! fstream_class FileWriter
npm ERR! fstream_finish_call utimes
npm ERR! code ENOENT
npm ERR! errno 34
npm ERR! fstream_stack /usr/lib/nodejs/fstream/lib/writer.js:305:19
npm ERR! fstream_stack Object.oncomplete (evalmachine.<anonymous>:107:15)
npm ERR! ref@1.3.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ref@1.3.2 install script.
npm ERR! This is most likely a problem with the ref package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls ref
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.3.0
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "chip-io"
npm ERR! cwd /home/chip/Documents/sandbox/nodetest01
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/chip/Documents/sandbox/nodetest01/npm-debug.log
npm ERR! not ok code 0
chip@chip:~/Documents/sandbox/nodetest01$

RyanOC, 2016-05-30:
I think I know what it is now. I'm going to try adding a symlink of node. Right now I only have nodejs as a command.

RyanOC, 2016-05-30:
That didnt work. Here is my new error log from the terminal...

chip@chip:~/Documents/sandbox/nodetest01$ npm install chip-io
npm WARN package.json nodetest01@0.0.0 No description
npm WARN package.json nodetest01@0.0.0 No repository field.
npm WARN package.json nodetest01@0.0.0 No README data
-
> ref@1.3.2 install /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ref
> node-gyp rebuild

make: Entering directory '/home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ref/build'
  CXX(target) Release/obj.target/binding/src/binding.o
In file included from ../src/binding.cc:7:0:
../node_modules/nan/nan.h:330:47: error: 'REPLACE_INVALID_UTF8' is not a member of 'v8::String'
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
binding.target.mk:82: recipe for target 'Release/obj.target/binding/src/binding.o' failed
make: *** [Release/obj.target/binding/src/binding.o] Error 1
make: Leaving directory '/home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ref/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.3.0
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ref
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 

> ffi@2.0.0 install /home/chip/Documents/sandbox/nodetest01/node_modules/chip-io/node_modules/ffi
> node-gyp rebuild

gyp ERR! UNCAUGHT EXCEPTION 
gyp ERR! stack Error: ENOENT, no such file or directory
gyp ERR! stack     at process.cwd (/usr/lib/nodejs/graceful-fs/polyfills.js:8:19)
gyp ERR! stack     at Object.exports.resolve (path.js:309:52)
gyp ERR! stack     at configure (/usr/share/node-gyp/lib/configure.js:26:23)
gyp ERR! stack     at Object.self.commands.(anonymous function) [as configure] (/usr/share/node-gyp/lib/node-gyp.js:66:37)
gyp ERR! stack     at run (/usr/share/node-gyp/bin/node-gyp.js:72:30)
gyp ERR! stack     at process._tickCallback (node.js:419:13)
gyp ERR! System Linux 4.3.0
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"

/usr/lib/nodejs/graceful-fs/polyfills.js:8
    cwd = origCwd.call(process)
                  ^
Error: ENOENT, no such file or directory
    at process.cwd (/usr/lib/nodejs/graceful-fs/polyfills.js:8:19)
    at errorMessage (/usr/share/node-gyp/bin/node-gyp.js:119:28)
    at issueMessage (/usr/share/node-gyp/bin/node-gyp.js:125:3)
    at process.<anonymous> (/usr/share/node-gyp/bin/node-gyp.js:109:3)
    at process.emit (events.js:95:17)
    at process._fatalException (node.js:272:26)
npm ERR! ref@1.3.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ref@1.3.2 install script.
npm ERR! This is most likely a problem with the ref package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls ref
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.3.0
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "chip-io"
npm ERR! cwd /home/chip/Documents/sandbox/nodetest01
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/chip/Documents/sandbox/nodetest01/npm-debug.log
npm ERR! not ok code 0
chip@chip:~/Documents/sandbox/nodetest01$ node -v
v0.10.29
chip@chip:~/Documents/sandbox/nodetest01$ nodejs -v
v0.10.29

RyanOC, 2016-05-30:
Ok, sorry for any time wasted. I realized that I didn't install the debian nodejs. As soon as I corrected this mishap, I was able to install chip-io and johnny-five successfully. Hopefully my errors help someone else.

Also, thanks very much for taking the time to put together this package!

Ryan

closed 2016-05-30 by RyanOC

sandeepmistry, 2016-05-31:
@RyanOC no worries, great to see that you found a solution.

Let me know if the instructions in the read me can be clarified, so others don't run into this.