sandeepmistry/node-chip-io #6: LCD pin access

ugate, 2016-07-07:
[LCD pin access for faster GPIO](http://docs.getchip.com/chip.html#expanding-gpio) than what's available on XIO/PCF8574A

sandeepmistry, 2016-07-13:
@ugate do you have more info. on this? Could you please also elaborate on what your issue/request is?

All I see is:

> If you don’t need to drive an LCD, you can use those pins for more, faster GPIO if you want to. These are the pins numbered 18-40 on U13 and 27-40 on U14 to act as GPIO to increase the number of available GPIO pins. Documentation on this process is forthcoming!

urish, 2016-07-14:
According to this post:

https://bbs.nextthing.co/t/muxing-chip-gpios/300/8

It is possible to create a DTO (Device Tree Overlay) that will configure the pins differently, possibly gaining access to additional GPIOs. 

Here is a thread worth following where the participants document their effort trying to get custom DTOs to get SPI functionality:

https://bbs.nextthing.co/t/self-help-group-to-understand-device-tree-overlays/5817

urish, 2016-07-14:
Actually, there seems to be a python library that does the trick:
https://github.com/xtacocorex/CHIP_IO

Also, this thread includes a complete GPIO pin-mapping for many of the pins, including the LCD ones:
https://bbs.nextthing.co/t/solved-gpio-pin-numbers/2931

sandeepmistry, 2016-07-15:
@urish thanks for the info! My CHIP's been in a drawer for a few months, I didn't realize there was a new 4.4 release that doesn't enable the LCD pins.

There's already experimental support for the LCD pins built-in. Just run with `sudo node <file>.js` and use a pin name from `lib/pins.json` which has a supported mode entry.

It currently accesses the registers directly by opening `/dev/mem` and memory mapping. I remember starting to look at PWM and LRADC via a similar technique, but can't find my progress at the moment. The cool thing is this approach doesn't require DTO, just running with`sudo`. I think `CHIP_IO` require `sudo` access to export the pins anyways, is this correct?

urish, 2016-07-15:
@sandeepmistry apparently, no DTO is needed. I could easily get an LED connected to one of the LCD pins to blink by running:

    var led = new five.Led('LCD-D3');
    led.blink(500);

I also tried with some other pins (such as `CSID0`), with a positive result.

So I think that this issue can be closed, and perhaps we can update the README to reflect the fact that these pins are available for use.

p.s. any special reason for putting this CHIP in the drawer? Did you discover some other cool boards to work with?

urish, 2016-07-15:
By the way, not sure why - but it seems like it runs well even without `sudo`, at least for the blink example above.

ugate, 2016-07-15:
I can also confirm LCD pin access as well. I overlooked the memory mapping. It would be great to update the README to indicate what pins are available for use as @urish suggested?

@sandeepmistry low res PWM and LRADC would be great. Should we open another issue for these? And while we're at it, what are your thoughts for SPI and UART (or @urish)?

sandeepmistry, 2016-07-17:
@urish @ugate thanks for trying this out!

> By the way, it seems like some of the pins are allowed even with sudo, including STATUS and the XIO pins.

Yes, any of the current pins listed in the read me only require `i2c` permission. The LCD ones will required sudo or root.

> I think a better behavior would be to throw a JS Exception in this case.

Agreed, we can detect this because the open will fail in `lib/allwinner-r8.js`. Do you have time to submit a PR for this? I think an exception should only be thrown if one of the R8 pins is being accessed and you do not have permission.

> It would be great to update the README to indicate what pins are available for use as @urish suggested?

Anyone have time to submit a pull request for this? `lib/pins.json` can be used to generate an updated table for pins that have `supportedModes.length > 0`. We should also add a new column "Requires sudo or root" which is set if `chip === 'R8'`.

> low res PWM and LRADC would be great. Should we open another issue for these?

I've pushed some changes to support this to master :) Please try them out ...

> And while we're at it, what are your thoughts for SPI and UART (or @urish)?

For SPI: let's start an issue to discuss this.

For UART: I think [node-serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport) will work out of the box for `/dev/ttyS0` and `/dev/ttyS1`. However, one gotcha is ttyS0 has a bash shell connected, ttyS1 has the Bluetooth adapter connected. I'm not too keen to making a JS driver for ttyS2. A DTO is probably better suited.

> p.s. any special reason for putting this CHIP in the drawer? Did you discover some other cool boards to work with?

I wouldn't say cooler, but I've been playing with the Nordic nRF51 and nRF52 (see [arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5)). As well as the BBC micro:bit (see [node-bbc-microbit-io](https://github.com/sandeepmistry/node-bbc-microbit-io) and [node-bbc-microbit](https://github.com/sandeepmistry/node-bbc-microbit)). Plus also new Raspberry Pi 3 and Zero.

urish, 2016-07-17:
great, I will have at exception throwing PR :-)

I also started playing with nRF51 last week, though I don't have a programmer so I am using a Raspberry PI 3 with OpenOCD as a programmer. Still haven't managed to get the Arduino IDE build process produce hex file that would actually work on the chip, but I will have a look at your arduino-nRF5 repo - perhaps it could shed some light on this :)

p.s. thanks for creating all these awesome repos - whenever I look for anything related to BLE, I somehow end-up at your github :+1:

urish, 2016-07-17:
On a different note: 

If you want to access the LCD/CSID pins without using `sudo` to run node.js, here is how to do it:

    sudo chown chip /dev/mem
    sudo setcap cap_sys_rawio+ep `which node`

a word of warning though - this also means that your `chip` user gets unlimited power as it gets direct WRITE access to all system memory.

sandeepmistry, 2016-07-18:
We could add a note in the read me about this in an "advanced configuration" section. However, I think for most users using `sudo` is the easiest. What do you think?

urish, 2016-07-21:
Yup, that makes total sense. Also, I think that the permissions on /dev/mem are reset whenever the system is rebooted...

ugate, 2016-07-21:
@sandeepmistry @urish I opened [CHIP-linux issue #30](/NextThingCo/CHIP-linux/issues/30) to to add a `gpio` group to address the `sudo` requirement. I think this is how RPi addresses this with `/dev/gpiomem`.

sandeepmistry, 2016-07-23:
I've update the pin table in the read me in c4156551e53d599fa783acfc876c8b4c35bbaca2. As well as published v2 to npm.

closed 2016-07-23 by sandeepmistry

urish, 2016-07-24:
Well done @sandeepmistry, looks great!