sandeepmistry/node-chip-io #9: i2c-1 pins not exposed on PocketCHIP

bbx10, 2016-07-19:
The pocketCHIP exposes the i2c-2 pins but not i2c-1 so changing the default to i2c-2 would it possible to use J5 i2c on the pocketCHIP.

Also /dev/i2c-1 is not available on Headless 4.4.

```
chip@chip44:~$ ls -l /dev/i2c*
crw-rw---- 1 root i2c 89, 0 Jul 19 07:17 /dev/i2c-0
crw-rw---- 1 root i2c 89, 2 Jul 19 07:17 /dev/i2c-2
```
Changing chip-io.js to use /dev/i2c-2 works OK for HTU21D and BMP180 i2c sensors.

```
  var i2c = new I2C(2, address);
  var i2c = new I2C(2, address);
```

sandeepmistry, 2016-07-20:
Hi @bbx10,

Thanks for the heads up on this, could you please submit pull request for this change? I'll release  v2.0 later this week.

bbx10, 2016-07-20:
@sandeepmistry 

Oops, please hold on this change. I am getting garbled results now but it was working last night. I hope I have not fried my sensor boards.

bbx10, 2016-07-20:
I found the bmp180 produces wildly varying results even when used on CHIP GUI 4.3 and i2c-1. The bmp180 fails in the same way on GUI 4.3 i2c-2 and Headless 4.4 i2c-2.

The htu21d produces correct temperature values in all 3 cases above. The humidity part of the sensor may be damaged because it works only sometimes. I do not currently see a pattern to the failure.

My conclusion: Changing from bus 1 to 2 is OK.

ugate, 2016-07-21:
@bbx10  I'm going to runs some tests using a PCA9685 over i2c using i2c-2 on v4.4. I'll report back the results here. I've also reviewed the johnny-five API and found that we should be adding/updating the bus number via options passed into five class derivatives like `Servo` [constructors](http://johnny-five.io/api/servo/#general-options). I've verified that any options passed into classes like `Servo` are passed into `Board.i2cConfig` as an object (not milliseconds as the docs indicate- see below) using [the current stable release of johnny-five](https://www.npmjs.com/package/johnny-five). I'll submit a pull request once I have tested the fix for this thoroughly.

[image]
 
This will allow us to do things like:
```js
var chipio = require('chip-io');
var board = new five.Board({
  io: new chipio(),
  timeout: 9999999
});
board.on("ready", function() {
  console.log("Connected");
  // Initialize the servo instance
  var a = new five.Servo({
    address: 0x40,
    controller: "PCA9685",
    pin: 0,
    bus: 2 // define chip-io i2c bus here
  });
  var b = new five.Servo({
    address: 0x40,
    controller: "PCA9685",
    range: [0, 180],
    pin: 1,
    bus: 2 // define chip-io i2c bus here
  });
  var degrees = 0;
  a.to(degrees);
  b.to(degrees);
});
```

sandeepmistry, 2016-07-23:
PR #10 from @bbx10 fixes this. Also #12 from @rwaldron will use the bus passed into `i2cConfig`, and defaults to bus 2.

I'll update the read me accordingly.

closed 2016-07-23 by sandeepmistry