sandeepmistry/node-chip-io #11: PCA9685 over chip-io I2C not working

ugate, 2016-07-20:
I tested a PCA9685 with chip-io (master branch) and noticed that nothing was happening using the [johnny-five example](http://johnny-five.io/examples/servo-PCA9685/). It looks like [errors are just consumed in i2c.js](/sandeepmistry/node-chip-io/blob/master/lib/chip-io.js#L203) so there's no way of knowing unless debugging (headless chip, used `node --inspect` flag with devtools from remote PC). I noticed that `chip-io.js` uses a hard-coded **1** for the bus value in `/dev/i2c-1` [here](/sandeepmistry/node-chip-io/blob/master/lib/chip-io.js#L193) and [here](/sandeepmistry/node-chip-io/blob/master/lib/chip-io.js#L245), but a `ls /dev` only shows `i2c-0` and `i2c-2` (Linux 4.4.11-ntc). I'm not sure why there's no option to pass the buss number in the [johnny-five API](http://johnny-five.io/api/board/#i2cconfigmilliseconds) for the `Board.i2c` functions? How else can boards that support multiple I2C bus lines (like CHIP) determine which one to use?

rwaldron, 2016-07-21:
Platform specific `bus` property support needs to be handled by the IO Plugin—all Johnny-Five does is forward the options passed to the component class constructor, through to the the `i2cConfig(...)` call.

rwaldron, 2016-07-21:
What version of Node.js is on the Chip?

rwaldron, 2016-07-21:
nvm, I see that this project points to the nodesource install instructions—very smart.

rwaldron, 2016-07-21:
I just found this: https://github.com/xtacocorex/CHIP_IO/issues/11

rwaldron, 2016-07-21:
@ugate [This should allow you to specify the bus](https://github.com/sandeepmistry/node-chip-io/pull/12) as shown in the following:

```js
var a = new five.Servo({
  bus: 1,
  address: 0x40,
  controller: "PCA9685",
  pin: 0,
});
```

ugate, 2016-07-21:
@rwaldron Thanks for the update! I noticed the same thing on #9. What threw me off is the docs that state that `Board.i2cConfig([milliseconds])` takes [an optional millisecond parameter rather than an object of options](http://johnny-five.io/api/board/#i2cconfigmilliseconds). Should the docs be updated to reflect an options object instead?

closed 2016-07-23 by sandeepmistry

sandeepmistry, 2016-07-23:
@ugate it would be awesome if you could submit a pull request to the Johnny Five docs your `Board.i2cConfig([milliseconds])` observation.

ugate, 2016-07-23:
@sandeepmistry [done](/rwaldron/io-plugins#11) :+1: