sandeepmistry/node-chip-io #13: Why setting all pins to 0 an initialisation?

Torsten85, 2016-07-24:
Hi there,

just one question. In this line:
https://github.com/sandeepmistry/node-chip-io/blob/cf615dd7440c6afb6bfe4b4b5e84980816e7caa0/lib/pcf8574a.js#L21
You reset all pins to 0. 

This will cause my relays (active low) to open on startup. This causes all sorts of problems in my setup, so I have to override the open function to prevent this (this doesn't look pretty).

Could you possible make this behavior configurable or remove this line?

Thanks in advance.
Torsten

sandeepmistry, 2016-07-25:
Hi @Torsten85,

Maybe we can read the current state of the PCF8574A and use that as starting point?

One note, this could also upset users who's devices are active high depending on what the default state of the PCF8574A is on boot.

Torsten85, 2016-07-26:
I though the pin state is read from the filesystem and has not to be set first... so set the starting point at all? 
Couldn't you just simply don't initialize the pins so they remain in the current state?

sandeepmistry, 2016-08-22:
The PCF8574A is a bit weird, there's no way to know the read/write mask when first starting up.

Maybe you can submit a PR to further discuss your idea?

wiktorschmidt, 2016-09-11:
Another approach would be to allow users to specify initial state. Similarly to what chip-gpio does (https://www.npmjs.com/package/chip-gpio) with low/high options:

> direction - A string specifying whether the GPIO should be configured as an input or output. The valid values are: 'in', 'out', 'high', and 'low'. 'high' and 'low' are variants of 'out' that configure the GPIO as an output with an initial level of high or low respectively.

In my experience all pins are set to high at CHIP boot.

sandeepmistry, 2016-09-17:
> Another approach would be to allow users to specify initial state. 

@wiktorschmidt do you have a proposal for how this could be integrated with the Johnny-Five API?

J5's PCF8574 code currently also sets all pins to output mode and low:

https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L387-L388

wiktorschmidt, 2016-09-17:
Not really. Again - it's a weird choice as according to http://www.ti.com/product/PCF8574

> At power on, the I/Os are high.

I would argue it shouldn't change the power on state at init without reason.

sandeepmistry, 2016-09-17:
@wiktorschmidt please submit a pull request of what you have in mind, then we can discuss further :)

sandeepmistry, 2016-10-10:
I'm closing this for now, please open a pull request for ideas on implementing the suggestions.

closed 2016-10-10 by sandeepmistry