sandeepmistry/node-chip-io #14: Button not working?

Torsten85, 2016-07-26:
It seems like I'm doing something wrong, but the button example doesn't work for me.

I'm on 4.4 headless and plugged two wires in GND and XIO-P1. After starting the button demo I don't see any output when a connect the wires. 
What am I missing?

sandeepmistry, 2016-07-27:
@Torsten85 try a circuit like this: http://johnny-five.io/examples/button/ - however use 3.3V instead of 5V.

Torsten85, 2016-07-27:
@sandeepmistry thanks. That circuit did work. I know is far beyond the scope of this package, but could you tell how one would connect this multi axis analog button / joystick? 

http://thingnovation.com/media/catalog/product/cache/3/image/06bc8a3647af2c2623702854802975ac/p/s/psp-2-axis-analog-thumb-game-joystick-module-3v-5v-for-arduino-psp-1.jpg

rwaldron, 2016-07-27:
Connect the `vrx` and `vry` each to an analog input. Then: http://johnny-five.io/api/joystick/ and http://johnny-five.io/examples/joystick/

sandeepmistry, 2016-07-27:
Unfortunately the C.H.I.P. only has one LRADC (6-bit) port.

So, you'll need to use an external ADC board. J5 has built in support for the [LIS3DH](http://johnny-five.io/examples/expander-LIS3DH/) and [PCF8591](http://johnny-five.io/examples/expander-PCF8591/).

@rwaldron do you have any other suggestions?

Torsten85, 2016-07-28:
is it possible to use this joystick in a digital way to get only infos about the direction (right -> yes /no), not how much it is pushed?

rwaldron, 2016-07-28:
> is it possible to use this joystick in a digital way to get only infos about the direction (right -> yes /no), not how much it is pushed?

Joysticks are variable resistance sensors, which means they produce a value that represents a variably resisted voltage reading, as a result of applied physical change. This isn't a boolean/logical state change (yes or no, high or low, on or off). If you want "yes/no", then the appropriate component is a button or a switch, otherwise you'll have to write a bunch of mechanics to filter the wide range of values. We can provide guidance for that, but a change of hardware should be considered.

sandeepmistry, 2016-08-21:
I'm closing this, since the original topic/question has been solved.

Feel free to continue the discussion though.

closed 2016-08-21 by sandeepmistry