sandeepmistry/node-chip-io #16: Is serial port access supported?

nkolban, 2016-08-14:
I was looking at the following page and was surprised to see that Serial/UART is flagged as not supported?

http://johnny-five.io/platform-support/#chip

Is this in fact correct?  If yes, is there a technical reason?  If not, and you just haven't had time ... I'd be more than happy to collaborate if welcome.

sandeepmistry, 2016-08-21:
Hi @nkolban,

I'm not too sure what the Serial/UART flag means, because there is no J5 API for this from what I can see. Maybe it means if the the remote board needs to be connected via a Serial port. @rwaldron do you have any clarifications for this?

In any case, I think [node-serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport) should work out of the box. So, there's no need to add a custom version. What do you think?

monteslu, 2016-08-22:
A serial passthrough was added to the firmata protocol  and the firmata.js client recently

https://github.com/firmata/protocol/blob/master/serial.md
https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L1687-L1864

This allows us to do things such as connect a GPS to an arduino's RX/TX pins and read it from johnny-five.

I'm not sure yet what that means for IO classes that run node on-board.

rwaldron, 2016-08-31:
It's included in IO-Plugins https://github.com/rwaldron/io-plugins#minimum-api-requirements

sandeepmistry, 2016-09-05:
@rwaldron thanks for the clarification.

When I started `boards-io` was missing the `serial*` API's (https://github.com/achingbrain/board-io/commit/2854e2ee57de86b0d77ee704e282ec01794c6b40).

> If not, and you just haven't had time ... I'd be more than happy to collaborate if welcome.

@nkolban that would be great if you or someone else could add this.

nkolban, 2016-09-06:
Let me start reading up on the internals of chip-io and make sure I'm up to the task.  I wrote up about chip-io in a new book available here ... https://leanpub.com/chip

I'm very much a fan of the project so if I can help, I will.

sandeepmistry, 2016-09-06:
Hi @nkolban,

Sounds good, I think you we need to glue the following together: https://github.com/achingbrain/board-io/commit/2854e2ee57de86b0d77ee704e282ec01794c6b40 + [node-serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport) (and figure out which `/dev/tty*` device maps to the broken out pins on the CHIP).

Please let us know if you plan to work on the feature, if not we can leave it open for someone else to contribute.

sandeepmistry, 2016-10-10:
It doesn't look like anyone's working on this, so it's open for pull requests.