sandeepmistry/node-chip-io #17: autoscroll() not working

sumarouno, 2016-09-18:
I have lcd 16x2 with i2c backpack (PCF8574T). I run this code:
```
var five = require('johnny-five');
var chipio = require('chip-io');

var board = new five.Board({
  io: new chipio()
});

board.on('ready', function() {
        var lcd = new five.LCD({controller: "PCF8574T", address: 0x27, bus: 2, rows: 2, cols: 16});
        lcd.cursor(0, 0);
        lcd.autoscroll().print("Bloop").print("Bleep");
});

```
My lcd only display BloopBleep and a blinking cursor.
Thanks

rwaldron, 2016-09-23:
There are a number of issues emerging with I2C LCD devices and CHIP, I will post back here and on the other as soon as I learn more

sandeepmistry, 2016-09-24:
Hi @sumarouno,

Could you please also try this code with PR #19?

sumarouno, 2016-09-24:
Yes sure. I have tried that code. But still same. I got BleepBoop and blinking cursor.
Thanks

sandeepmistry, 2016-10-02:
Hi @sumarouno,

Could you also provide debug logs? `DEBUG=chip-io node <file>.js`

I'm interested to see what i2c methods are being called.

sumarouno, 2016-10-03:
> DEBUG=chip-io node <file>.js

hi, @sandeepmistry . Nice to meet you again. I will try your suggestion but I'm at my office right now. I will give my report after I try at home.
Thanks

sumarouno, 2016-10-03:
@sandeepmistry , here the debug output:
```bash
1475495767000 Device(s) C.H.I.P.  
1475495767033 Connected C.H.I.P.  
1475495767109 Repl Initialized  
>>   chip-io i2cConfig +0ms Options { controller: 'PCF8574T', address: 39, bus: 2, rows: 2, cols: 16 }
  chip-io i2cWrite +232ms 39 null 0
  chip-io i2cWrite +17ms 39 null 48
  chip-io i2cWrite +7ms 39 null 52
  chip-io i2cWrite +11ms 39 null 48
  chip-io i2cWrite +9ms 39 null 48
  chip-io i2cWrite +8ms 39 null 52
  chip-io i2cWrite +5ms 39 null 48
  chip-io i2cWrite +9ms 39 null 48
  chip-io i2cWrite +3ms 39 null 52
  chip-io i2cWrite +3ms 39 null 48
  chip-io i2cWrite +3ms 39 null 32
  chip-io i2cWrite +2ms 39 null 36
  chip-io i2cWrite +2ms 39 null 32
  chip-io i2cWrite +9ms 39 null 36
  chip-io i2cWrite +3ms 39 null 32
  chip-io i2cWrite +10ms 39 null 132
  chip-io i2cWrite +3ms 39 null 128
  chip-io i2cWrite +4ms 39 null 4
  chip-io i2cWrite +2ms 39 null 0
  chip-io i2cWrite +3ms 39 null 196
  chip-io i2cWrite +3ms 39 null 192
  chip-io i2cWrite +4ms 39 null 4
  chip-io i2cWrite +2ms 39 null 0
  chip-io i2cWrite +3ms 39 null 20
  chip-io i2cWrite +3ms 39 null 16
  chip-io i2cWrite +6ms 39 null 4
  chip-io i2cWrite +2ms 39 null 0
  chip-io i2cWrite +3ms 39 null 100
  chip-io i2cWrite +11ms 39 null 96
  chip-io i2cWrite +4ms 39 null 8
  chip-io i2cWrite +6ms 39 null 140
  chip-io i2cWrite +3ms 39 null 136
  chip-io i2cWrite +3ms 39 null 12
  chip-io i2cWrite +3ms 39 null 8
  chip-io i2cWrite +3ms 39 null 12
  chip-io i2cWrite +3ms 39 null 8
  chip-io i2cWrite +2ms 39 null 220
  chip-io i2cWrite +3ms 39 null 216
  chip-io i2cWrite +13ms 39 null 77
  chip-io i2cWrite +2ms 39 null 73
  chip-io i2cWrite +6ms 39 null 45
  chip-io i2cWrite +2ms 39 null 41
  chip-io i2cWrite +8ms 39 null 109
  chip-io i2cWrite +3ms 39 null 105
  chip-io i2cWrite +2ms 39 null 205
  chip-io i2cWrite +3ms 39 null 201
  chip-io i2cWrite +2ms 39 null 109
  chip-io i2cWrite +15ms 39 null 105
  chip-io i2cWrite +2ms 39 null 253
  chip-io i2cWrite +6ms 39 null 249
  chip-io i2cWrite +6ms 39 null 109
  chip-io i2cWrite +3ms 39 null 105
  chip-io i2cWrite +2ms 39 null 253
  chip-io i2cWrite +3ms 39 null 249
  chip-io i2cWrite +3ms 39 null 125
  chip-io i2cWrite +2ms 39 null 121
  chip-io i2cWrite +3ms 39 null 13
  chip-io i2cWrite +2ms 39 null 9
  chip-io i2cWrite +5ms 39 null 77
  chip-io i2cWrite +2ms 39 null 73
  chip-io i2cWrite +3ms 39 null 45
  chip-io i2cWrite +2ms 39 null 41
  chip-io i2cWrite +6ms 39 null 109
  chip-io i2cWrite +2ms 39 null 105
  chip-io i2cWrite +5ms 39 null 205
  chip-io i2cWrite +2ms 39 null 201
  chip-io i2cWrite +2ms 39 null 109
  chip-io i2cWrite +3ms 39 null 105
  chip-io i2cWrite +3ms 39 null 93
  chip-io i2cWrite +2ms 39 null 89
  chip-io i2cWrite +3ms 39 null 109
  chip-io i2cWrite +2ms 39 null 105
  chip-io i2cWrite +3ms 39 null 93
  chip-io i2cWrite +2ms 39 null 89
  chip-io i2cWrite +3ms 39 null 125
  chip-io i2cWrite +3ms 39 null 121
  chip-io i2cWrite +5ms 39 null 13
  chip-io i2cWrite +3ms 39 null 9

```

sumarouno, 2016-10-03:
I also try autoscroll() for my mega2560 and my lcd 1602 with i2c backpack. What i got is the same thing, "BleepBloop" with blinking cursor at the end.
here my code for m mega:
```javascript
var five = require('johnny-five');

var boards = new five.Boards([
{
	id: "mega",
	port: "/dev/ttyUSB3"
},
]);

boards.on('ready', function() {
	var lcd = new five.LCD({
		controller: "PCF8574T", 
		address: 0x27, 
		bus: 2, 
		rows: 2, 
		cols: 16
	});

        lcd.cursor(0, 0);
        lcd.autoscroll().print("Bloop").print("Bleep");
});
```
thank you

sandeepmistry, 2016-10-03:
@sumarouno great idea to test with the Mega2560!

Since it doesn't seem specific to the CHIP + chip-io, I think we can close and move to the issue to the main [Johnny-Five repo](https://github.com/rwaldron/johnny-five).

closed 2016-10-03 by sumarouno