sandeepmistry/node-chip-io #18: installation issues

AnnaNican, 2016-09-23:
trying to install chip-io library with no success. any advice?

npm install johnny-five chip-io
npm ERR! Darwin 14.5.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "johnny-five" "chip-io"
npm ERR! node v5.4.1
npm ERR! npm  v3.10.8
npm ERR! code EBADPLATFORM

npm ERR! notsup Unsupported platform for chip-io@2.0.1: wanted {"os":"linux","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm ERR! notsup Valid OS:    linux
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   darwin
npm ERR! notsup Actual Arch: x64

nkolban, 2016-09-23:
The chip-io package is defined to be run on a Linux operating system.  It appears that you are running on a "darwin" operating system

https://en.wikipedia.org/wiki/Darwin_(operating_system) 

which I'm guessing means you are running on Mac OS X.  Unfortunately, I don't believe this Node.js package is supported on that platform.  Now whether that is because of a technical limitation or because the author of the package didn't have a Mac OS X to test with ... that I can't answer.   If we look at line 38 of `package.json` in the source of the project, we see a hard check for "linux".

rwaldron, 2016-09-23:
> Now whether that is because of a technical limitation or because the author of the package didn't have a Mac OS X to test with ... that I can't answer. 

This module is intentionally designed for running exclusively on [CHIP](https://getchip.com/) boards and nothing else.

sandeepmistry, 2016-09-24:
I've added a note to the read me for this in https://github.com/sandeepmistry/node-chip-io/commit/a2d8e13075e9a5b13f5fced11a77b83973716464.

@AnnaNican why are you trying to install on a Mac?

closed 2016-09-24 by sandeepmistry