sandeepmistry/node-chip-io #20: PCA9685 not working

ericterpstra, 2016-09-25:
Continued discussion from #8 ...

Can't get PCA9685 board with 2 small servos working.

Wiring:

CHIP => PCA9685
TWI2-SDA => SDA
TWI2-SCK => SCK
VCC 3v3 => VCC
GNC => GND

PCA9685 (no bridged jumpers - 0x40)
4.8V Battery => post
Fitech FS90R continuous servos on pins 7 & 8

Code

```
var five = require('johnny-five');
var chipio = require('chip-io');

var board = new five.Board({
  io: new chipio()
});

board.on('ready', function() {
  var statusLed = new chipio.StatusLed();

  var left = new five.Servo.Continuous({
    bus: 2,
    controller: "PCA9685",
    address: 0x40,
    pin: 8,
    specs: { speed: five.Servo.Continuous.speeds["@4.8V"] }
  });

  var right = new five.Servo.Continuous({
    bus: 2,
    controller: "PCA9685",
    address: 0x40,
    pin: 7,
    specs: { speed: five.Servo.Continuous.speeds["@4.8V"] }
  });

  left.sweep();
  right.sweep();
  statusLed.blink(500);
  this.repl.inject({ left:left, right:right});
});
```

Output from `i2cdetect -r 2`

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
```

Console output from running above code:

```
1474844239932 Device(s) C.H.I.P.
1474844239969 Connected C.H.I.P.
1474844240047 Repl Initialized
>> fs.js:786
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: ENXIO: no such device or address, write
    at Error (native)
    at Object.fs.writeSync (fs.js:786:20)
    at I2C.write (/home/chip/servotest/node_modules/chip-io/lib/i2c.js:21:6)
    at ChipIO.i2cWrite (/home/chip/servotest/node_modules/chip-io/lib/chip-io.js:210:9)
    at ChipIO.i2cWriteReg (/home/chip/servotest/node_modules/chip-io/lib/chip-io.js:221:8)
    at Expander.value (/home/chip/servotest/node_modules/johnny-five/lib/expander.js:632:17)
    at new Expander (/home/chip/servotest/node_modules/johnny-five/lib/expander.js:1868:10)
    at Function.Expander.get (/home/chip/servotest/node_modules/johnny-five/lib/expander.js:1907:10)
    at Servo.value (/home/chip/servotest/node_modules/johnny-five/lib/servo.js:24:35)
    at new Servo (/home/chip/servotest/node_modules/johnny-five/lib/servo.js:189:8)
```

Logic analyzer result when running the above code (had a friend collect this for me)

[i2c decode.csv.txt](https://github.com/sandeepmistry/node-chip-io/files/491952/i2c.decode.csv.txt)
[raw.csv.txt](https://github.com/sandeepmistry/node-chip-io/files/491953/raw.csv.txt)

sandeepmistry, 2016-09-25:
Could you also provide debug logs? `DEBUG=chip-io node <file>.js`

Have you tried to read and write to the controller using [`i2cget`](https://linux.die.net/man/8/i2cget) and [`i2cset`](https://linux.die.net/man/8/i2cset)?

ericterpstra, 2016-09-30:
Below is the debug output from `DEBUG=chip-io node index.js`

```
1475263881747 Device(s) C.H.I.P.
1475263881780 Connected C.H.I.P.
1475263881852 Repl Initialized
>>   chip-io pinMode +0ms 81 1
  chip-io i2cConfig +51ms Options {
  address: 64,
  controller: 'PCA9685',
  bus: 2,
  pwmRange: [ 450, 1850 ],
  frequency: 50 }
  chip-io i2cWriteReg +230ms 64 0 0
  chip-io i2cWrite +2ms 64 0 0
  chip-io i2cWriteReg +7ms 64 0 16
  chip-io i2cWrite +1ms 64 0 16
  chip-io i2cWriteReg +3ms 64 254 135
  chip-io i2cWrite +0ms 64 254 135
  chip-io i2cWriteReg +5ms 64 0 0
  chip-io i2cWrite +1ms 64 0 0
  chip-io i2cWriteReg +6ms 64 0 161
  chip-io i2cWrite +0ms 64 0 161
fs.js:786
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: ENXIO: no such device or address, write
    at Error (native)
    at Object.fs.writeSync (fs.js:786:20)
    at I2C.write (/home/chip/servotest/node_modules/chip-io/lib/i2c.js:21:6)
    at ChipIO.i2cWrite (/home/chip/servotest/node_modules/chip-io/lib/chip-io.js:210:9)
    at ChipIO.i2cWriteReg (/home/chip/servotest/node_modules/chip-io/lib/chip-io.js:221:8)
    at Expander.value (/home/chip/servotest/node_modules/johnny-five/lib/expander.js:632:17)
    at new Expander (/home/chip/servotest/node_modules/johnny-five/lib/expander.js:1868:10)
    at Function.Expander.get (/home/chip/servotest/node_modules/johnny-five/lib/expander.js:1907:10)
    at Servo.value (/home/chip/servotest/node_modules/johnny-five/lib/servo.js:24:35)
    at new Servo (/home/chip/servotest/node_modules/johnny-five/lib/servo.js:189:8)
```

So it should be apparent by now that I have no idea what I'm doing, but I did find some i2cset commands on the internet that managed to spin a servo on channel 0.

https://www.raspberrypi.org/forums/viewtopic.php?p=105657

I plugged in the following, and the servo started spinning:

```
Making PCA9685 go to sleep
/usr/sbin/i2cget -y 2 0x40 0
0x00

set sleep bit while preserving other bits and ensuring that restart bit is set to 0 when writing to register
/usr/sbin/i2cset -y 2 0x40 0 16
Setting frequency to 60 Hz. Calcualting prescale value prescale=round( 25000000/(4096 * frequency)) - 1 from frequency as 101(PCA9685 clock is at 25MHz or 25000000 Hz)
/usr/sbin/i2cset -y 2 0x40 254 101
/usr/sbin/i2cset -y 2 0x40 0 0x00

usleep 5000
/usr/sbin/i2cset -y 2 0x40 0 128
Setting Channel 0 PWM on at 0 step, off at 150 step in 0 to 4095 steps at 60 Hz
/usr/sbin/i2cset -y 2 0x40 6 0
/usr/sbin/i2cset -y 2 0x40 7 0
/usr/sbin/i2cset -y 2 0x40 8 150
/usr/sbin/i2cset -y 2 0x40 9 0
Setting Channel 0 PWM on at 3 step, off at 0 step in 0 to 4095 steps at 60 Hz
/usr/sbin/i2cset -y 2 0x40 6 3
/usr/sbin/i2cset -y 2 0x40 7 0
/usr/sbin/i2cset -y 2 0x40 8 0
/usr/sbin/i2cset -y 2 0x40 9 0
```

sandeepmistry, 2016-10-02:
Ok, a few suggestions to try out:

1) Only declare one servo in your J5 code. Also, are you sure it's a 4.8V continuous servo?

2) Map the debug output to `i2cset` commands and run them to see what happens

```
/usr/sbin/i2cset -y 2 64 0 0
/usr/sbin/i2cset -y 2 64 0 16
/usr/sbin/i2cset -y 2 64 254 135
/usr/sbin/i2cset -y 2 64 0 0
/usr/sbin/i2cset -y 2 64 0 161
```

I'm wondering if the last value is too high or something.

The `ENXIO` error code maps to "device did not respond" (http://stackoverflow.com/q/38400940/2020087).

ericterpstra, 2016-10-04:
So the only thing that makes it go is this:

```
/usr/sbin/i2cset -y 2 0x40 8 150
```
I've looked over the docs for the [PCA9685](https://cdn-shop.adafruit.com/datasheets/PCA9685.pdf), the [servo](https://cdn-shop.adafruit.com/product-files/2442/FS90R-V2.0_specs.pdf), and i2cset. I can't make sense of how the servo pulse-width range (700μs~2300μs) equates to the i2c prescale, steps, and registers, and how it all plugs into i2cset.  All that really makes sense to me is that registers 6, 7, 8, and 9 are for pin 0.

:/

rwaldron, 2016-10-04:
This is not related to your issue, but you can remove the `specs` property altogether: it isn't used for anything <3 I know it appears in the example, but that's just an old piece of code that never got updated when the `specs` feature was punted. I've updated and it and the github/website/examples will update shortly. 

- https://github.com/rwaldron/johnny-five/commit/f8f4e6ac782acc04e772bea4ebe8f3c8666d2fda
- https://github.com/rwaldron/johnny-five/commit/f8f4e6ac782acc04e772bea4ebe8f3c8666d2fda

As for prescaling, steps and registers: this is all board agnostic and Johnny-Five does that work for you. It's effectively all the same as what you've written here: https://github.com/sandeepmistry/node-chip-io/issues/20#issuecomment-250834936 in the last portion of the comment. 

@sandeepmistry does https://github.com/fivdi/i2c-bus run on CHIP? If it does, I would suggest implementing the I2C capabilities with that. Generally, the advice is to migrate all IO Plugins (where applicable) to use that. cc @fivdi

rwaldron, 2016-10-04:
Here's the implementation of I2C capabilities for pcDuino, using i2c-bus: https://github.com/rwaldron/pcduino-io/blob/master/lib/pcduino-io.js#L268-L385

ugate, 2016-10-05:
@rwaldron I'm testing out some updates that use `ioctl` [messaging bundles](sandeepmistry/node-chip-io/issues/15#issuecomment-250995558) which essentially allows us to send multiple I2C reads into a single `ioctl` call. This eliminates a lot of overhead when multiple reads are being processed. The implementation code is also much simpler than the API used to write directly to the `i2c-#` files. All r/w byte, r/w block, etc. can be bundled into a single `ioctl` call, even when reading from different addresses and/or registers. And due to address/register message bundling, it also eliminates the need to call `ioctl(fd, I2C_SLAVE_FORCE, address)`. And to take things further, pending read buffers are pre-constructed during queuing for `tick` interval processing so when it's time to actually call `ioctl` there's no build operations to perform.

There's still some pending tests that I'd like to setup before submitting a PR, but I've tested a few devices like the demanding *runner* LCD + I2C backback setup and it seems to be working well. It's still using `ffi`, but if you think there would be some value added to convert it to a very simple `nan` impl we could do that as well.

fivdi, 2016-10-05:
> @sandeepmistry does https://github.com/fivdi/i2c-bus run on CHIP? If it does, I would suggest implementing the I2C capabilities with that. Generally, the advice is to migrate all IO Plugins (where applicable) to use that. cc @fivdi

I ordered a CHIP recently and will test i2c-bus on it when it ships. If someone wants to give it a try, a quick test would be `npm install i2c-bus`. Then take a look at [i2cscan.js](https://github.com/fivdi/i2c-bus/blob/v1.1.2/example/i2cscan.js) in the example directory. When run, `i2cscan` should output the same information as `i2cdetect -y -r 1`. Note that `i2cscan` is hard-coded to scan bus 1. To scan a different bus [this line](https://github.com/fivdi/i2c-bus/blob/v1.1.2/example/i2cscan.js#L7) would need to be modified to pass the appropriate bus number to the openSync method.

ericterpstra, 2016-10-05:
@fivdi Looks like it works.  Output below (for i2c-2):

```
chip@chip:~/i2c-bus/example$ sudo node i2cscan.js
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
```

Not sure if it helps or not, but also changed i2cdetect.js to open i2c-2, and this was the output:

```
Functionalities implemented by /dev/i2c-2:
I2C                              yes
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               yes
SMBus Block Write                yes
SMBus Block Read                 no
SMBus Block Process Call         no
SMBus PEC                        yes
I2C Block Write                  yes
I2C Block Read                   yes
```

bbx10, 2016-11-27:
```
  chip-io i2cWriteReg +230ms 64 0 0
  chip-io i2cWrite +2ms 64 0 0
  chip-io i2cWriteReg +7ms 64 0 16
  chip-io i2cWrite +1ms 64 0 16
```

There are a number of writes to register 0 but there is a bug (and fix) for this case. Try installing chip-io from git and see if #27 helps.

bbx10, 2016-11-29:
@ericterpstra @sandeepmistry 
I got a PCA9685 board working by installing chip-io from github. In other words, #27 fixes this issue.

sandeepmistry, 2016-12-02:
@bbx10 thanks for trying this out letting us know.

Maybe you can share more info on how you wired things up so the @ericterpstra can take a look?

In any case, I'll close this for now since it is functional.

closed 2016-12-02 by sandeepmistry

bbx10, 2016-12-02:
There is nice message on the CHIP forum that shows how to wire the board and run a simple test with an LED.

https://bbs.nextthing.co/t/johnny-five-chip-io-and-pca9685/12386

sandeepmistry, 2016-12-05:
@bbx10 that looks great, thanks for sharing the link!

It's strange that the latest from Github is needed, let me know if you think I should publish a new version to npm. #27 should already be included in the version published to npm currently.

bbx10, 2016-12-05:
#27 is dated Oct 30 but version 2.0.2 was released on Oct 9 so publishing a new version should be done. I also compared chip-io.js from node-chip-io versus node_modules/chip-io/src. The npm installed version does not have #27.

sandeepmistry, 2016-12-05:
@bbx10 ok, sorry for the confusion.

I'll publish a new version later today.

sandeepmistry, 2016-12-05:
@bbx10 version 2.1.0 has been published.

bbx10, 2016-12-06:
@sandeepmistry Thanks! I installed chip-io and johnny-five using npm and PCA9685 works fine.