sandeepmistry/node-chip-io #21: PWM0 appears to implement range incorrectly

jmaxxz, 2016-10-07:
Tested using a led between pwm0 (18) and GND

led.off(); //turned on the led
led.on(); //turned off the led
led.brightness(0); // turned on led
led.brightness(255) // turned off led

LED api: http://johnny-five.io/api/led/

jmaxxz, 2016-10-07:
Example with output from oscilloscope
```javascript
const five = require('johnny-five');
const chipio = require('chip-io');

var board = new five.Board({
  io: new chipio()
});

board.on('ready', function() {
  var led = new five.Led(18);
  led.on();
});

```

[image]

rwaldron, 2016-10-07:
Looks inverted! I wonder if the PWM channel polarity needs to be flipped?

sandeepmistry, 2016-10-09:
I've flipped the PWM polarity in https://github.com/sandeepmistry/node-chip-io/commit/8afcb5389418fabea50e192e750e74cf4ec165ff. @jmaxxz please try the change out via `npm install sandeepmistry/node-chip-io`.

closed 2016-10-09 by sandeepmistry

jmaxxz, 2016-10-09:
Will try it out tonight.

jmaxxz, 2016-10-10:
Works much better!  However, on is not fully on, the waveform in the scope screenshot has been inverted with this change.

sandeepmistry, 2016-10-10:
@jmaxxz thanks for trying it out!

> However, on is not fully on, the waveform in the scope screenshot has been inverted with this change.

Could you please open a new issue to track this. For 0 and 255, I think we need to disable PWM and switch to normal GPIO mode ...