sandeepmistry/node-chip-io #22: Use i2c-bus module for I2C layer

sandeepmistry, 2016-10-09:
As discussed by @rwaldron  in https://github.com/sandeepmistry/node-chip-io/issues/20#issuecomment-251506686

> @sandeepmistry does https://github.com/fivdi/i2c-bus run on CHIP? If it does, I would suggest implementing the I2C capabilities with that. Generally, the advice is to migrate all IO Plugins (where applicable) to use that. cc @fivdi

and https://github.com/sandeepmistry/node-chip-io/issues/20#issuecomment-251510047

> Here's the implementation of I2C capabilities for pcDuino, using i2c-bus: https://github.com/rwaldron/pcduino-io/blob/master/lib/pcduino-io.js#L268-L385

We should look into swapping out the current [node-ffi](https://github.com/node-ffi/node-ffi) based I2C layer with  @fivdi's [i2c-bus](https://github.com/fivdi/i2c-bus) module.

fivdi, 2016-10-10:
If you want, I can take a look at this when the CHIPs I ordered a while back arrive.

sandeepmistry, 2016-10-10:
@fivdi thanks, that would be great!

If anyone else starts to work on this, please post an update here so we avoid duplicate efforts.

ugate, 2016-10-10:
@sandeepmistry @fivdi I've already begun work on an implementation that utilizes a more generic/flexible use of `I2C_RDWR` versus `I2C_FUNCS` that's used in `i2c-bus`. This will vastly reduce boilerplate `nan` logic and allow consecutive reads to be called in a single `ioctl` call rather than an `ioctl` call for each read.

fivdi, 2016-10-15:
@ugate yes, I agree, the amount of boilerplate logic required when implementing native addons can be high. However, the real issue, in my opinion, isn't necessarily the amount of boilerplate required but the amount of knowledge required to understand and implement that boilerplate. Not only is knowledge of nan required. Knowledge of the inner workings of Node.js, v8, nan, uv, async processing, C++, one or more Operating Systems, etc... is also required.

If the addon you're implementing will allow multiple operations to be performed with a single call to ioctl, how can this be utilized by the current Johnny-Five io-plugin architecture? Or would the current io-plugin architecture need to change?

ugate, 2016-10-15:
@fivdi The current architecture can remain the same. Read operations are queued/pre-built and take advantage of the *async* nature of the `i2cRead*` interface. Each call to `i2cRead*` is combined into a single buffer until they are processed via the board's sampling interval, at which point a single synchronous call to `ioctl` is made and each corresponding *handler* is processed using values set by `ioctl`.  This extricates transactions from consecutive stop/start conditions as well as consolidating calls to `ioctl` that would otherwise be repeated for separate reads. Writes on the other hand are constrained by the J5 io-plugin architecture, but could potentially be called in a similar manner if there were a use case to warrant doing so.

patonz, 2016-11-16:
Just for info: can i use "BMP180" Thermometer like :  ```
temp = new five.Thermometer({
    controller: "BMP180",
    freq: 5000
  });
```
ive connected this one on the i2c bus and after run ive have this message:

> (node:1572) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 2): Error: ENXIO: no such device or address, write
> (node:1572) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

tell me if the i2c is ready to use on chip using chip-io or ive missed something :(

sandeepmistry, 2016-11-16:
@patonz please open a new issue and share a photo of your wiring.

fivdi, 2016-12-28:
The CHIPs that I ordered a while back arrived recently and I hope to work on this in January.

@ugate are you working on an alternative implementation?

sandeepmistry, 2017-01-03:
@fivdi cool, @bbx10 might be looking at, but I haven't heard anything yet.

ugate, 2017-01-04:
@fivdi I am working on an alternative implementation that basically uses `ioctl` messages. Part of my testing has been adding a lot of other features to `chip-io`.

sandeepmistry, 2017-01-16:
Closed via #37.

closed 2017-01-16 by sandeepmistry