sandeepmistry/node-chip-io #23: chip's onboard button

norgeous, 2016-10-10:
Hello,
I do not own any loose electronic components to do proper GPIO with.
I see that it's possible to access the onboard status led with:
https://github.com/sandeepmistry/node-chip-io/blob/master/examples/blink-status.js

But is it also possible to access the tiny onboard button as seen in the very bottom left (beside LCD-HSYCN)?
[image]

It would be great to be able to do the same as:
https://github.com/sandeepmistry/node-chip-io/blob/master/examples/led-button.js
but with the onboard button and status led.

sandeepmistry, 2016-10-10:
Hi @norgeous,

Good question. I believe that button is to manage the AXP290, so if you hold it the chip will power off, then you can turn the CHIP back on by pressing it again..

I would suggest you get a button or simulate a button with a wire connected to 3.3V or GND for testing.

norgeous, 2016-10-10:
Thanks!
Yes, holding the button does force a shutdown but the button is not exclusively for this.
It can also be used for short presses and accessed in bash using these commands
```bash
/usr/sbin/i2cget -f -y 0 0x34 0x4a           # get button value, 0x01=off and 0x03=on
/usr/sbin/i2cset -f -y 0 0x34 0x4a 0x2       # resets button value to 0x01
```
Here is a one line test program for your SSH:
```bash
while true; do /usr/sbin/i2cget -f -y 0 0x34 0x4a; /usr/sbin/i2cset -f -y 0 0x34 0x4a 0x2; sleep 2; done;
```
If the status led access code `new chipio.StatusLed();` uses `i2cset` behind the scenes, can't this onboard button be accessed in the same way? I'm new to all this GPIO/i2c stuff...

I guess the real tricky part is making the button 'realtime',
accessing it without the polling (every 2 seconds in my bash example above) would be awesome.
Linked up to be event driven would be perfect! 

Given the above info maybe reopen this bug?

sandeepmistry, 2016-10-10:
Ah, interesting!

We can add a entry to the `tick` loop which happens every 19ms like for the internal temp and battery: https://github.com/sandeepmistry/node-chip-io/blob/master/lib/axp209.js#L41-L53

Do you think you can work on a PR for this?

norgeous, 2016-10-10:
I think I could, and had literally just found that exact file moments ago...
I would start by adding:
```
var BUTTON = 0x4a;
```
but idk where i'd go from there...
I think it would be way faster if you could implement this (pretty please).

norgeous, 2016-10-10:
Just a little reiteration on my first post to save you some time...

The button acts kind of like a toggle, once pressed `/usr/sbin/i2cget -f -y 0 0x34 0x4a` will always return `0x03`.

Un-toggling is achieved with `/usr/sbin/i2cset -f -y 0 0x34 0x4a 0x2` and now the `i2cget` command will now return `0x01`

sandeepmistry, 2016-10-16:
@norgeous please see #25 and try it out + provide your feedback :)

norgeous, 2016-10-16:
Yes, certainly!
Will check its working shortly...
Thanks!

norgeous, 2016-10-16:
I just tested it on a fresh stock flash (4.4 headless) with git (2.1.4) , node (v6.8.1) and build-essential installed, with the following commands:
```
git clone https://github.com/sandeepmistry/node-chip-io.git -b button
cd node-chip-io
npm i
node examples/onboard-button.js
```
and it looks like it works perfectly!
I get `down` then `up` instantly logged to the console on each press.

If I create a PR for a new file `examples/status-onboard-button.js` (fits current name structure) which would trigger the onboard led with the onboard button, would you merge it?

These new features make me to think that a few of the terms in the `examples/` folder should be redone to clarify things... perhaps `status` should be `onboard-led` but will leave that for now... perhaps a separate PR.

Thank you for your time spent on this improvement to the project!

norgeous, 2016-10-16:
actually, it's not working as I expected it to...
pressing the button does give `down` then `up` to the console,
but both are displayed on button release.

Button down gives nothing, but release gives both `down` then `up` immediately.
Is this intended?

norgeous, 2016-10-16:
also, FYI a little further testing reveals:

power:
1. holding the button for 6 full seconds turns off the chip.
2. Press for 1 full second to switch back on.

short press:
1. a 'short press' appears to be less than 2 seconds
2. holding the button for more than 2 sec, but less than 6 sec does nothing (0x34 0x4a remains unchanged).

times are approximate.

norgeous, 2016-10-16:
I just discovered something else by running:

`while true; do /usr/sbin/i2cget -f -y 0 0x34 0x4a; /usr/sbin/i2cset -f -y 0 0x34 0x4a 0x03; sleep 0.5; done;`

With this script we are setting as `0x03` instead of `0x02`, now both long and short presses can be read:

short = `0x02`
long = `0x01`

In both cases setting it as `0x03` returns `0x00` - which is way more intuitive.

Thing is, I don't think the short press bit is set until the button is released, so I don't think you will be able to fix the
`onboardButton.on('down'...`
as the button will not tell us once the button is pressed (only on release).

Conversely, for long press the bit is set to `0x01` immediately without releasing it and there is no bit change on release.

Would be nice to include these features also, or just remove the `on down` function.

Thanks @sandeepmistry!

norgeous, 2016-10-16:
as it is a non-standard button and will not behave with `down` and `up`,
perhaps you could code some custom events such as:
```
onboardButton.on('short', function() {
  console.log('short');
});
onboardButton.on('long', function() {
  console.log('long');
});
```
idk if this is possible.

sandeepmistry, 2016-10-23:
@norgeous thanks for trying it out and the feedback!

> If I create a PR for a new file examples/status-onboard-button.js (fits current name structure) which would trigger the onboard led with the onboard button, would you merge it?

Yes, that would be cool.

> Is this intended?

Not intended but it's a limitation of AXP290, it doesn't given you the press event just the release as you mentioned.

What if we just remove the press event from the example?

sandeepmistry, 2016-10-30:
@norgeous ping ...

norgeous, 2016-10-31:
> What if we just remove the press event from the example?

Yes, I think that's a good compromise.

I will prepare a PR for the other example, shortly (I hope).

sandeepmistry, 2016-11-01:
Closed via #25.

closed 2016-11-01 by sandeepmistry