sandeepmistry/node-chip-io #27: error in chip-io.js that breaks i2C on some chips [fix in comments]

neilhand, 2016-10-29:
Not sure how to submit an actual fix on github, but i found an issue that i wanted to report. 

The issue relates to any I2C device that has a register with an address of zero and requires the address as part of the command. The specific one that was an issue for me was an i2c servo controller, but i am sure there are more. 

In the chip-io.js file on line 208 and 252 there is a malformed comparison. 

The comparison is 

```
if (register) {
...
}
```
However this matched not only the register being set to null as intended, but also the register address being 0x00.

The fix is to change the comparison to explicitly check for null.

```
if (register != null) {
}
```

I hope this helps someone else since it was a pain to find, and then only through resorting to an i2c protocol analyzer and comparison between different platforms.

sandeepmistry, 2016-10-30:
Hi @neilhand,

Thanks for bringing this up! I've pushed your suggested change:  https://github.com/sandeepmistry/node-chip-io/commit/f713760452a02f7c80e19212e225b14e1d1c1e85 - let me know if it looks ok, then I can publish a new version to npm with the fix.

In the future, you can always submit a pull request and I'd be happy to help you out with any questions you had on how to do this.

https://yangsu.github.io/pull-request-tutorial/

closed 2016-10-30 by sandeepmistry

neilhand, 2016-11-01:
> Sandeep Mistry <mailto:notifications@github.com>
> October 30, 2016 at 7:27 AMvia Postbox
> f713760

looks good. i'll look over the tutorial its been something ive been 
meaning to do for a long time.

/Neil