sandeepmistry/node-chip-io #28: HTU21D fails with version 2.0.2 but works with 2.0.1

bbx10, 2016-11-24:
HTU21D works with chip-io@2.0.1 but fails with version 2.0.2. BMP180 works with both.

```
1479955744648 Device(s) C.H.I.P.  
1479955744685 Connected C.H.I.P.  
1479955744786 Repl Initialized  <<< Freezes here with 2.0.2 
>> 53.642791748046875 %
53.635162353515625 %
```

The node process freezes. Ctrl-C does not work. The process must be killed from another shell.

Code used.

```
var five = require('johnny-five');
var chipio = require('chip-io');
var board = new five.Board({
  io: new chipio()
});

board.on("ready", function() {
  var hygrometer = new five.Hygrometer({
    controller: "HTU21D"
  });

  hygrometer.on("change", function() {
    console.log(this.relativeHumidity + " %");
  });
});
```

sandeepmistry, 2016-11-26:
Hi @bbx10,

Unfortunately I don't have a `HTU21D` device on hand to debug with. There were some big changes to I2C in #19.

Maybe the debug logs will help? `DEBUG=chip-op node <files>.js`

bbx10, 2016-11-26:
Thanks, the debug output is useful. It appears the I2C reads succeed but the data/events never reach the top level code.

Example code reading temperature and humidity.

```
var five = require("johnny-five");
var chipio = require('chip-io');
var board = new five.Board({io: new chipio()});

board.on("ready", function() {
  var multi = new five.Multi({
    controller: "HTU21D"
  });

  multi.on("change", function() {
    console.log("Thermometer");
    console.log("  celsius           : ", this.thermometer.celsius);
    console.log("  fahrenheit        : ", this.thermometer.fahrenheit);
    console.log("  kelvin            : ", this.thermometer.kelvin);
    console.log("--------------------------------------");

    console.log("Hygrometer");
    console.log("  relative humidity : ", this.hygrometer.relativeHumidity);
    console.log("--------------------------------------");
  });
});
```

```
chip-io@2.0.1
chip@chip1:~/j5$ DEBUG=chip-io node htu21dmulti.js
1480185181898 Device(s) C.H.I.P. 
1480185181933 Connected C.H.I.P. 
1480185182034 Repl Initialized 
>>   chip-io i2cConfig Options {
  controller: 'HTU21D',
  freq: undefined,
  board: undefined,
  address: 64,
  delay: 50000 } +0ms
  chip-io i2cWrite 64 null 254 +44ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +84ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +20ms
Thermometer
  celsius           :  27.23
  fahrenheit        :  81.01
  kelvin            :  300.38
--------------------------------------
Hygrometer
  relative humidity :  53.574127197265625
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms
Thermometer
  celsius           :  27.25
  fahrenheit        :  81.05
  kelvin            :  300.4
--------------------------------------
Hygrometer
  relative humidity :  53.574127197265625
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +23ms
Thermometer
  celsius           :  27.26
  fahrenheit        :  81.07
  kelvin            :  300.41
--------------------------------------
Hygrometer
  relative humidity :  53.566497802734375
--------------------------------------
Thermometer
  celsius           :  27.26
  fahrenheit        :  81.07
  kelvin            :  300.41
--------------------------------------
Hygrometer
  relative humidity :  53.566497802734375
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms
```

The datasheet says the worst case for humidity conversion is 50 ms. Does this means sync read will block for up to 50 ms? The debug show a time of 84 ms which I do not understand.

```
chip-io@2.0.2
chip@chip1:~/j5$ DEBUG=chip-io node htu21dmulti.js
1480185077629 Device(s) C.H.I.P. 
1480185077665 Connected C.H.I.P. 
1480185077741 Repl Initialized 
>>   chip-io i2cConfig Options {
  controller: 'HTU21D',
  freq: undefined,
  board: undefined,
  address: 64,   
  delay: 50000 } +0ms
  chip-io i2cWrite 64 null 254 +44ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +19ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +71ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +19ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +16ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +48ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +22ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +48ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +52ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +58ms
```

The I2C reads appear to be working but the data/events never reach the top level code. Also Ctrl-C does not work. The node process must be killed from another shell.

I will get some I2C bus traces to see if the I2C operations are really working or not.

bbx10, 2016-11-27:
The I2C bus trace looks fine to me. Using version 2.0.2.

[chipio_202_htu21df.csv.txt](https://github.com/sandeepmistry/node-chip-io/files/614531/chipio_202_htu21df.csv.txt)

bbx10, 2016-11-27:
I think the problem is changing i2cReadOnce to sync results in a tight loop calling a blocking function which causes node to freeze.

The function readCycle calls itself from a callback function. When i2cReadOnce is async, it returns without calling readCycle. When the I2C I/O completes (10-50 ms later), the callback function is called which calls readCycle again.

When i2cReadOnce is sync, the callback function is called almost immediately after i2cReadOnce returns. Since the callback function calls readCycle, node freezes in a tight loop calling i2cReadOnce. Recall when used with the HTU21DF, i2cReadOnce/readSync can block for 50 ms or longer.

I confirmed this by changing (see listing below) `readCycle();` to `setTimeout(readCycle, 10);`

node no longer freezes with this additional delay. This is not a solution because this readCycle patterns repeats for many other devices.

One solution is to change I2C I/O back to async but apparently this breaks other devices. Or perhaps the i2cReadOnce must delay before calling its callback function. I am not sure what to do here.

```
imu.js
HTU21D:
  initialize:

        var readCycle = function() {
          // Despite the registers being back to back, the HTU21D
          // does not like when 5 bytes are requested, so we put
          // the two data sources on their own read channels.
          var isTemperatureCycle = cycle === 0;
          var register = isTemperatureCycle ? this.REGISTER.TEMPERATURE : this.REGISTER.HUMIDITY;

          io.i2cReadOnce(address, register, 2, function(data) {
            if (isTemperatureCycle) {
              computed.temperature = int16(data[0], data[1]);
            } else {
              computed.humidity = int16(data[0], data[1]);
            }

            if (++cycle === 2) {
              cycle = 0;
              this.emit("data", computed);
            }
            console.log('about to call readCycle');
            //readCycle();
            setTimeout(readCycle, 10);    // unfreezes node
          }.bind(this));
          console.log('i2cReadOnce returned');
        }.bind(this);
 
        readCycle();
```

rwaldron, 2016-12-01:
```js
setTimeout(readCycle, 10);    // unfreezes node
```

Can you make a PR to Johnny-Five with this fix?

sandeepmistry, 2016-12-02:
Interesting, @bbx10 @rwaldron has this been tested with any IO plugin's that use @fivdi's `i2c-bus` package?

Might make sense to look at swapping to use it as #22 suggests.

bbx10, 2016-12-02:
@rwaldron @sandeepmistry Ok, I submitted a PR. I can take a look at i2c-bus changes.

fivdi, 2017-01-06:
> Interesting, @bbx10 @rwaldron has this been tested with any IO plugin's that use @fivdi's i2c-bus package?

I haven't tested it as I don't have a HTU21D. However, I'm fairly confident that the process won't freeze with IO plugins using i2c-bus as all the IO plugins using i2c-bus that I'm aware of read from I2C devices asynchronously. They write to I2C devices synchronously. To be exact, the IO plugin `i2cRead` and `i2cReadOnce` methods are asynchronous and the `i2cWrite` and `i2cWriteReg` methods are synchronous in IO plugins that use i2c-bus.

rwaldron, 2017-01-09:
I can confirm @fivdi's statement above

sandeepmistry, 2017-01-16:
@bbx10 can you please try this again with master? I've merged #37 from @fivdi.

fivdi, 2017-01-22:
I now have a HTU21D and tried to reproduce the issue with master.

Here's htu21dmulti.js with a slight difference, the HTU21D is on bus 1:

```js
var five = require("johnny-five");
var chipio = require('chip-io');
var board = new five.Board({io: new chipio()});

board.on("ready", function() {
  var multi = new five.Multi({
    controller: "HTU21D",
    bus: 1
  });

  multi.on("change", function() {
    console.log("Thermometer");
    console.log("  celsius           : ", this.thermometer.celsius);
    console.log("  fahrenheit        : ", this.thermometer.fahrenheit);
    console.log("  kelvin            : ", this.thermometer.kelvin);
    console.log("--------------------------------------");

    console.log("Hygrometer");
    console.log("  relative humidity : ", this.hygrometer.relativeHumidity);
    console.log("--------------------------------------");
  });
});
```

And here's the output of `DEBUG=chip-io node htu21dmulti.js`:

```
chip@chip:~/chip-io-master$ DEBUG=chip-io node htu21dmulti.js
1485043436050 Device(s) C.H.I.P.  
1485043436085 Connected C.H.I.P.  
1485043436160 Repl Initialized  
>>   chip-io i2cConfig Options {
  controller: 'HTU21D',
  freq: undefined,
  board: undefined,
  bus: 1,
  address: 64,
  delay: 50000 } +0ms
  chip-io i2cWrite 64 254 undefined +178ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +19ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +88ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +25ms
Thermometer
  celsius           :  21.23
  fahrenheit        :  70.21
  kelvin            :  294.38
--------------------------------------
Hygrometer
  relative humidity :  28.442901611328125
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +46ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +21ms
Thermometer
  celsius           :  21.25
  fahrenheit        :  70.25
  kelvin            :  294.4
--------------------------------------
Hygrometer
  relative humidity :  28.450531005859375
--------------------------------------
Thermometer
  celsius           :  21.25
  fahrenheit        :  70.25
  kelvin            :  294.4
--------------------------------------
Hygrometer
  relative humidity :  28.450531005859375
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +45ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms
Thermometer
  celsius           :  21.27
  fahrenheit        :  70.29
  kelvin            :  294.42
--------------------------------------
Hygrometer
  relative humidity :  28.450531005859375
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +46ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms
Thermometer
  celsius           :  21.29
  fahrenheit        :  70.32
  kelvin            :  294.44
--------------------------------------
Hygrometer
  relative humidity :  28.450531005859375
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +46ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +17ms
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +17ms
Thermometer
  celsius           :  21.29
  fahrenheit        :  70.32
  kelvin            :  294.44
--------------------------------------
Hygrometer
  relative humidity :  28.458160400390625
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +46ms

(To exit, press ^C again or type .exit)
>>   chip-io i2cReadOnce 64 227 2 function () { [native code] } +65ms
Thermometer
  celsius           :  21.3
  fahrenheit        :  70.34
  kelvin            :  294.45
--------------------------------------
Hygrometer
  relative humidity :  28.458160400390625
--------------------------------------
  chip-io i2cReadOnce 64 229 2 function () { [native code] } +47ms
  chip-io i2cReadOnce 64 227 2 function () { [native code] } +18ms

1485043437139 Board Closing.  
chip@chip:~/chip-io-master$ 

```
I can't reproduce the issue. The process doesn't lock and can be terminated by hitting `ctrl+c` twice.

sandeepmistry, 2017-01-24:
@fivdi great, thanks for trying this out!

It sounds like we can close this issue and publish a new version to npm. What do you think?

fivdi, 2017-01-24:
>@fivdi great, thanks for trying this out!

You're welcome.

>It sounds like we can close this issue and publish a new version to npm. What do you think?

I agree.

sandeepmistry, 2017-01-25:
v2.1.1 has been published to npm 😄.

Thanks again for you help!

closed 2017-01-25 by sandeepmistry

fivdi, 2017-01-25:
I did a few tests with v2.1.1 from npm and everything worked as expected. There is a Johnny-Five specific issue with the HTU21D (https://github.com/rwaldron/johnny-five/issues/1278)

sandeepmistry, 2017-01-30:
@fivdi awesome!