sandeepmistry/node-chip-io #29: example onboard-button.js

c316, 2016-12-02:
If I try to use ``` var chipio = require('chip-io');``` I get an error

```1480649763702 Device(s) C.H.I.P.  
1480649763736 Connected C.H.I.P.  
1480649763812 Repl Initialized  
>> /home/chip/easy-button/index.js:10
  var onboardButton = new chipio.OnboardButton();
                      ^

TypeError: chipio.OnboardButton is not a constructor
    at Board.<anonymous> (/home/chip/easy-button/index.js:10:23)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
    at _combinedTickCallback (internal/process/next_tick.js:67:7)
    at process._tickDomainCallback (internal/process/next_tick.js:122:9)
    at Module.runMain (module.js:606:11)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
```

I think the example is missing some parts in the readme

chris-gunawardena, 2016-12-03:
Try this:
```
var Chip_io = require('chip-io');
var chip_io = new Chip_io();
var onboardButton = chip_io.OnboardButton();
```

sandeepmistry, 2016-12-05:
@c316 so the onboard button support is not published to npm yet, so unless you clone from Gtihub or install via `npm install sandeepmistry/node-chip-io` it will not be present.

Please try this and let me know what the result is. Closing this for now.

closed 2016-12-05 by sandeepmistry