sandeepmistry/node-chip-io #30: Faster GPIO

chris-gunawardena, 2016-12-03:
Currently a CSI GPIO write takes about [500us](https://gist.github.com/chris-gunawardena/c412cd80fa26fcf45aabccfc00e1e529) but using the c write() it should be around [2.8 microseconds](https://github.com/fordsfords/gpiotest). 

I see this uses [memcpy](https://github.com/sandeepmistry/node-chip-io/blob/master/lib/allwinner-r8.js#L198) as opposed to c [write](https://github.com/fordsfords/gpiotest/blob/master/csitest.c#L126).

Is it possible to increase the write speeds using c write() function similar to [this python CHIP GPIO lib](https://github.com/xtacocorex/CHIP_IO/blob/a9902d12678a44470ed120ef16c308b686420e44/source/event_gpio.c#L398) or this overhead coming from node/v8? Or is it just way I'm [testing](https://gist.github.com/chris-gunawardena/c412cd80fa26fcf45aabccfc00e1e529) this.

sandeepmistry, 2016-12-05:
@chris-gunawardena currently `node-ffi` is used instead of a native Node.js module. `node-ffi` is known not to have great performance, but it's a nice starting point.

Do you have time to experiment with switching to a native Node.js module?

chris-gunawardena, 2016-12-05:
@sandeepmistry I tried with node-gyp and write() and getting around 50us. 
[https://github.com/chris-gunawardena/chip-fast-gpio](https://github.com/chris-gunawardena/chip-fast-gpio)
I'll try again with node-gyp and  memcpy to see if it's any faster.

I also came across [this using /dev/mem](http://hackaday.com/2013/12/07/speeding-up-beaglebone-black-gpio-a-thousand-times/), not sure if it applies to CHIP as well but I couldn't see a /dev/mem

sandeepmistry, 2016-12-08:
Cool, can you try with the `mmap` approach used in this repo: https://github.com/sandeepmistry/node-chip-io/blob/master/lib/allwinner-r8.js

I expect it to be much faster than the write() approach ...

fivdi, 2016-12-28:
As a comparison, the [onoff performance-sync test](https://github.com/fivdi/onoff/blob/master/test/performance-sync.js) will do about 100,000 writes per second on any of CSID0 through CSID7 which is about 10us per write. It uses sysfs files and is pure JavaScript. I'll have to admit that quite a lot of time was spent on performance tuning. The same test on a Raspberry Pi 3 will do about 300,000 writes per second.

[pigpio](https://github.com/fivdi/pigpio) which uses the mmap approach has a [digital-write-performance test](https://github.com/fivdi/pigpio/blob/master/test/digital-write-performance.js) that will do about 2.3 million writes per second on a Rasoberry Pi 3.

@chris-gunawardena I'd imagine that chip-fast-gpio would be quite a bit faster if it cached the fd between calls to [gpio_write](https://github.com/chris-gunawardena/chip-fast-gpio/blob/master/chip-fast-gpio.cc#L75-L86).

ugate, 2017-01-04:
> I'd imagine that chip-fast-gpio would be quite a bit faster if it cached the fd between calls to gpio_write.

@fivdi That was something else that I did as part of #22 Each `fd` is cached for every IC component that `chip-io` uses when they are opened just before `ready` is emitted.

ugate, 2017-01-19:
I've noticed that there is a lot of `Object.defineProperty()`, `Object.defineProperties()` and [`Function.prototype.bind`](http://benediktmeurer.de/2015/12/25/a-new-approach-to-function-prototype-bind/) calls in J5/chip-io which suppresses V8 optimizations which could account for some of the slowness.

sandeepmistry, 2017-01-24:
@chris-gunawardena @ugate @fivdi I've opened PR #38 with some changes to use a native module for the GPIO connected to the AllWinner R8. Could you please try it out?

fivdi, 2017-01-27:
I did a few performance tests with `chip-io@2.1.1` and `sandeepmistry/node-chip-io#faster-gpio` using the following program which blinks an LED connected to CSID0 as fast as it can.
```js
'use strict';

var five = require('johnny-five');
var chipio = require('chip-io');

var board = new five.Board({
  io: new chipio()
});

board.on('ready', function() {
  var time,
    writesPerSecond,
    i;

  this.pinMode('CSID0', five.Pin.OUTPUT);

  time = process.hrtime();

  for (i = 1; i <= 100000; i += 1) {
    this.digitalWrite('CSID0', i % 2);
  }

  time = process.hrtime(time);
  writesPerSecond = Math.floor(i / (time[0] + time[1] / 1E9));

  console.log('writes per second', writesPerSecond);
});
```
chip-io@2.1.1 can do approximately 2,464 writes per second
sandeepmistry/node-chip-io#faster-gpio can do about 78,453 writes per second

So the faster-gpio branch is over 30 times faster than chip-io@2.1.1. This is a huge improvement.

As a comparison, the following program which does the same thing but uses `onoff` can do approximately 138,068 writes per second. `onoff` uses the GPIO sysfs interface. 
```js
'use strict';

var Gpio = require('onoff').Gpio,
  led = new Gpio(132, 'out'), // CSID0 is GPIO132
  time,
  writesPerSecond,
  i;

time = process.hrtime();

for (i = 1; i <= 1000000; i += 1) {
  led.writeSync(i % 2);
}

time = process.hrtime(time);
writesPerSecond = Math.floor(i / (time[0] + time[1] / 1E9));

console.log('writes per second', writesPerSecond);

led.unexport();
```

fivdi, 2017-01-27:
I'm not sure how safe [this code](https://github.com/sandeepmistry/node-chip-io/blob/faster-gpio/lib/allwinner-r8.js#L113-L122) is:
```js
  var datVal = this._readRegister(datReg);
  var pinMask = 1 << (register * 8 + index);

  if (value) {
    datVal |= pinMask;
  } else {
    datVal &= ~pinMask;
  }

  this._writeRegister(datReg, datVal);
```
The contents of a register is being read, modified and written. There may be a context switch to another process or the OS between `_readRegister` and `_writeRegister`. That other process or the OS may modify the exact same register.

sandeepmistry, 2017-01-30:
@fivdi thanks for looking into this!

Do you think we should use onoff for GPIO expose by the kernel and the naive module for other pins?

> The contents of a register is being read, modified and written. There may be a context switch to another process or the OS between _readRegister and _writeRegister. That other process or the OS may modify the exact same register

Good point, any suggestions on how to avoid this? We could add a set/clear register bit to the native module to make the condition less likely to occur ...

fivdi, 2017-01-30:
@sandeepmistry 

> Do you think we should use onoff for GPIO expose by the kernel and the naive module for other pins?

If onoff was an order of magnitude faster I'd say yes. But it's only 1.75 times faster so I'd say no. The only reason for mentioning onoff was because I somehow expected the mmap approach to be faster than it is. There are additional advantages to the mmap approach that are not available with onoff. For example, pull up/down resistors could be configured with the mmap approach, but not with onoff.

>> The contents of a register is being read, modified and written. There may be a context switch to another process or the OS between _readRegister and _writeRegister. That other process or the OS may modify the exact same register

> Good point, any suggestions on how to avoid this? We could add a set/clear register bit to the native module to make the condition less likely to occur ...

The easiest way to do this would be with set/clear registers, but the Allwinner R8 doesn't appear to have set/clear registers. The BCM2835 on Raspberry Pi has such registers (GPSET0, GPSET1, GPCLR0 and GPCLR1).