sandeepmistry/node-chip-io #32: missing johnny-five spec functionpingRead

MarioProjects, 2016-12-07:
Hello, I want to report an issue trying ro Connect CHIP with hc_sr04 and chip-io Johnny-five. I used the Code

var five = require("johnny-five");
var chipio = require('chip-io');

var board = new five.Board({
io: new chipio()
});

board.on("ready", function() {
var proximity = new five.Proximity({
controller: 'HCSR04',
pin: 'XIO-P0'
});

});

I got the error:

this.io.pingRead(settings, function(microseconds) {
^

TypeError: this.io.pingRead is not a function
at Proximity. (/home/chip/robotica/node_modules/johnny-five/lib/proximity.js:197:19)
at Proximity.Controllers.HCSR04.initialize.value (/home/chip/robotica/node_modules/johnny-five/lib/proximity.js:203:9)
at new Proximity (/home/chip/robotica/node_modules/johnny-five/lib/proximity.js:475:10)
at Board. (/home/chip/robotica/johnny_five/hc_sr04_solo.js:10:19)
at emitNone (events.js:72:20)
at Board.emit (events.js:166:7)
at nextTickCallbackWith0Args (node.js:420:9)
at process._tickDomainCallback (node.js:390:13)
at Function.Module.runMain (module.js:443:11)
at startup (node.js:139:18)

I connected GND to GND, Trig and Echo to XIO-P0 and VCC to 3.3

rwaldron, 2016-12-07:
We need to revisit the docs for the `HCSR04` controller. This is only available when programming for an Arduino that's running the PingFirmata variant. `pingRead` is not supportable on platforms without high resolution timing necessary for synchronous digital write and `pulseIn` sequences. To use an HCSR04 sensor with any non-Arduino, you'll need to use a microcontroller "backpack" as an intermediary, as shown here: http://johnny-five.io/examples/proximity-hcsr04-i2c/, just replace the UNO with the CHIP. The firmware for the `HCSR04I2CBACKPACK`, which runs on the Nano, is available here: https://github.com/ajfisher/nodebots-hcsr04#i2c-backpack (includes instructions)

MarioProjects, 2016-12-07:
mmm sorry if It sounds greedy or something like this... I am a beginner and I only want to know if will be able in future versions of chip-io this feature because I am really interested in concéntrate hc_sr04 and pca9685 only in chip. Thanks for your work really, only I want to know :)

MarioProjects, 2016-12-07:
I searched a lot about this, dont think that I only want fast responses 😙

rwaldron, 2016-12-07:
>  I am a beginner and I only want to know if will be able in future versions of chip-io this feature

As I explained previously, this feature cannot be implemented on boards that aren't capable of timing sensitive `pulseIn` operations.

MarioProjects, 2016-12-08:
mmm ok so

sandeepmistry, 2016-12-18:
@rwaldron thanks for helping out with this!

Do you think this module should have a nicer error message when someone tries to use `.io.pingRead`?

MarioProjects, 2016-12-18:
I think that It is possible to run hc_sr04 on CHIP, maybe not with the same presition that un anuncio Arduino but...

sandeepmistry, 2016-12-18:
@MarioProjects 

> I think that It is possible to run hc_sr04 on CHIP, maybe not with the same presition that un anuncio Arduino

In that case please submit a pull request with your suggestion.

Please note, I don't have HCSR04 to test with.

closed 2016-12-18 by sandeepmistry

rwaldron, 2016-12-19:
@sandeepmistry I've always just recommended implementing functions that can't be supported, but have them throw an exception with useful explanation.

sandeepmistry, 2017-01-16:
@rwaldron good suggestion! I've added it in https://github.com/sandeepmistry/node-chip-io/commit/d0b82bdd606d9a29efbff381a11b9cce6c556771.