sandeepmistry/node-chip-io #40: Example issue?

gregwinn, 2017-02-01:
I am running the LED example: https://github.com/sandeepmistry/node-chip-io/blob/master/examples/led.js

But when I start it I get the following:

```
1485974497418 Device(s) C.H.I.P.  
1485974497454 Connected C.H.I.P.  
1485974497528 Repl Initialized  
>> 
```

I have to ^C to close it:

```
>> 
1485974539082 Board Closing.
```

Running as root.

Any idea whats going wrong here??

fivdi, 2017-02-01:
Everything is working fine. What you're seeing here is two features provided by Johnny-Five; debug output and REPL (the Node.js read eval print loop.) Both features are enabled by default and documented [here](http://johnny-five.io/api/board/#parameters). They can be disabled as follows:
```js
var five = require('johnny-five');
var chipio = require('../index'); // or require('chip-io');

var board = new five.Board({
  io: new chipio(),
  repl: false,
  debug: false
});

board.on('ready', function() {
  // Create an LED on the XIO-P0 pin
  var led = new five.Led('XIO-P0');

  // Blink every half second
  led.blink(500);
});
```

gregwinn, 2017-02-01:
Ok, very cool! Now, while that is going on my LED was not blinking. I will try without the debug and repl and let you know.

gregwinn, 2017-02-01:
OK, now when I run it it just hangs.

```
var five = require('johnny-five');
var chipio = require('chip-io');

var board = new five.Board({
    io: new chipio(),
    repl: false,
    debug: false
});

board.on('ready', function() {
    // Create an LED on the XIO-P0 pin
    var led = new five.Led('XIO-P0');

    // Blink every half second
    led.blink(500);
});
```

fivdi, 2017-02-01:
Hmm... the LED should blink irrespective of whether or not REPL and/or debug are activated. And the program will look like it's hanging but it's actually running and blinking the LED (or at least it should).

Are you 100% sure that your LED is connected to `XIO-P0` on the GPIO header?

fivdi, 2017-02-01:
And that the LED is connected up correctly? i.e., the short connection on the LED is connected to 0V and the long connection on the LED is connected to `XIO-P0` through a resistor?

gregwinn, 2017-02-01:
Yes but i have included a photo just in case I am missing something.

Setup image: https://s3.amazonaws.com/thinaer-static-images/IMG_20170201_132937.jpg

I have tested the LED with the 3v header and it will turn on so I know the LED is good.

fivdi, 2017-02-01:
I see that you're taking a no risk no fun approach and working without a resistor :)

Did you test the LED by connecting the yellow wire to 3V?

gregwinn, 2017-02-01:
LOL, resistors are for people that know what they are doing!

Anyway, yes I did test the circuit by connecting yellow to 3v to make sure it was all working. The LED lights up when connected to 3v.

gregwinn, 2017-02-01:
I am running the node script as `root`.  Any issue with that?

```
root@chip:~# node gpio.js 
1485977971798 Device(s) C.H.I.P.  
1485977971837 Connected C.H.I.P. 
```

fivdi, 2017-02-01:
For XIO-P0 root privileges shouldn't be needed but it shouldn't result in any problems either.

fivdi, 2017-02-01:
What's the output of the following command?
```
uname -a
```

gregwinn, 2017-02-01:
`Linux chip 4.4.13-ntc-mlc #1 SMP Thu Sep 8 19:54:49 UTC 2016 armv7l GNU/Linux`

fivdi, 2017-02-01:
Just a minute ...

fivdi, 2017-02-01:
Try the following command as root:
```
root@chip:~# cd /sys/class/gpio/
root@chip:/sys/class/gpio# echo 1013 > export 
root@chip:/sys/class/gpio# cd gpio1013
root@chip:/sys/class/gpio/gpio1013# echo out > direction 
root@chip:/sys/class/gpio/gpio1013# echo 1 > value 
root@chip:/sys/class/gpio/gpio1013# 
```
It should turn the LED on. If it doesn't something is wrong.

gregwinn, 2017-02-01:
Ok, something is wrong :(

No errors but the LED did not light up

gregwinn, 2017-02-01:
I am going to try on a fresh out of the bag CHIP

fivdi, 2017-02-01:
Maybe you should try with a resistor this time. Do you have one? Anything from 100 ohms to 1000 ohms?

fivdi, 2017-02-01:
Or try with a resistor on the same CHIP first?

gregwinn, 2017-02-01:
I now have a resistor in place, it did not work on CHIP 1, I am installing on CHIP 2 now

fivdi, 2017-02-01:
Try the command line stuff in `/sys/class/gpio` first.
An make sure the output of `uname -a` is `Linux chip 4.4.13-ntc-mlc #1 SMP Thu Sep 8 19:54:49 UTC 2016 armv7l GNU/Linux`.

gregwinn, 2017-02-01:
Ok, This is a brand new CHIP I did not install anything on it yet... Just connected it to wifi then ran:

```
root@chip:~# uname -a
Linux chip 4.4.13-ntc-mlc #1 SMP Thu Sep 8 19:54:49 UTC 2016 armv7l GNU/Linux
root@chip:~# cd /sys/class/gpio/
root@chip:/sys/class/gpio# echo 1013 > export
root@chip:/sys/class/gpio# cd gpio1013
root@chip:/sys/class/gpio/gpio1013# echo out > direction
root@chip:/sys/class/gpio/gpio1013# echo 1 > value
root@chip:/sys/class/gpio/gpio1013#
```

LED did not light up. I have included another photo that shows the full setup.
https://s3.amazonaws.com/thinaer-static-images/IMG_20170201_141448.jpg

fivdi, 2017-02-01:
That's a 220 ohm resistor, correct?

fivdi, 2017-02-01:
The setup looks good to me.

gregwinn, 2017-02-01:
So this is clearly not a `node-chip-io` npm issue. I am going to post in the forums on CHIP to see if anyone has an idea. I really appreciate your support with this! I will follow up on this ticket with the solution and a link to the CHIP forum topic.

THANK YOU!!!

fivdi, 2017-02-01:
Can you connect the yellow wire to CSID0 instead of XIO-P0 and then try the following from the command line?
```
root@chip:~# cd /sys/class/gpio/
root@chip:/sys/class/gpio# echo 132 > export 
root@chip:/sys/class/gpio# cd gpio132
root@chip:/sys/class/gpio/gpio132# echo out > direction 
root@chip:/sys/class/gpio/gpio132# echo 1 > value 
```

gregwinn, 2017-02-01:
That works!

gregwinn, 2017-02-01:
Works on both the original one CHIP 1 and the new CHIP with nothing on it CHIP 2

fivdi, 2017-02-01:
Ok. Asking at the forum is a good idea. Maybe there's something special with XIO-P0 through XIO-P7 when it comes to driving LEDs. It does work for me although the LED is a lot dimmer on XIO-P0 than it is on CSID0.

gregwinn, 2017-02-01:
Very good, I have updated the following like in the led example

`var led = new five.Led('CSID0');`

And it all works! I will follow up with this ticket after I get some useful information from the CHIP form.

Thank you very much!

fivdi, 2017-02-01:
>Thank you very much!

Your welcome.

And please do update this ticket with any useful information.

fivdi, 2017-02-01:
So, driving LEDs with XIO-P0 through XIO-P7 is special. XIO-P0 through XIO-P7 are connected to pins on a PCF8574a 8-bit I/O expander. The specialty is explained here: http://forum.arduino.cc/index.php?topic=19046.0. If LEDs are connected as suggested in the linked post they are nice and bright when turned on but there is an unfortunate side effect. Calling `led.off()` will turn the LED on and calling `led.on()` will turn the LED off.

sandeepmistry, 2017-02-02:
@fivdi thanks for helping out!

That lines up with what @interested1 mentioned to me in an email a while ago:

>Glad I asked how you wired it up. My suspicion is that your LED is dim because you didn't wire the LED correctly for C.H.I.P.. Unlike Arduino, C.H.I.P. needs to be wired so that it goes high when it sinks current. Try wiring the LED's cathode to the GPIO pin and place a resistor between the anode and 5V.

I've never had problems with the LED not working, it's only been dim for me with the `XIO-P*` pins.

fivdi, 2017-02-02:
> @fivdi thanks for helping out!

@sandeepmistry  No problem :)

> I've never had problems with the LED not working, it's only been dim for me with the `XIO-P*` pins.

I haven't had problems with the LED not working either but they have always been dim on `XIO-P*` when connected incorrectly, extremely dim.

@gregwinn were you by any chances working under conditions where is was very bright and where the LED may actually have been blinking so dimly that it didn't appear to be blinking? If so, maybe the blinking would be visible if the room was darker? Or could you try it with a different LED to see if it works, maybe an LED with a different color?

gregwinn, 2017-02-02:
@fivdi I will check, I am pretty sure it was not working but based on this information It is possible. I will let you know. 

Thanks @sandeepmistry 
> I've never had problems with the LED not working, it's only been dim for me with the XIO-P* pins.

gregwinn, 2017-02-06:
Ok, after a few days... I really did not get too much help on the forums. This post here led me to the conclusion that using `XIO-PX` is not a great choice for LED's but using `CSIDX` works just the same. I am able to use a button to trun an LED on with the button connected to a `XIO-PX` pin and the LED on a `CSIDX` pin.

@fivdi and @sandeepmistry have been a great help and I am able to use `node-chip-io`! I am going to close this ticket.

Thanks for all the wonderful support!

closed 2017-02-06 by gregwinn

fivdi, 2017-02-06:
@gregwinn using CSIDX rather than XIO-PX for LEDs sounds like a good decision to me.