sandeepmistry/node-chip-io #42: Add battery percent and current

norgeous, 2017-02-16:
So it currently supports the battery voltage via:

`var BAT_VOLT_MSB_REGISTER = 0x78;` (in `/lib/axp209.js`)

Looking at [battery.sh](https://raw.githubusercontent.com/KoljaWindeler/CHIP-hwtest/master/chip-hwtest/bin/battery.sh) I believe support for the battery's percent and amps can be done via commands like:
```
BAT_VOLT_MSB=$(i2cget -y -f 0 0x34 0x78)
BAT_VOLT_LSB=$(i2cget -y -f 0 0x34 0x79)

BAT_IDISCHG_MSB=$(i2cget -y -f 0 0x34 0x7C)
BAT_IDISCHG_LSB=$(i2cget -y -f 0 0x34 0x7D)

BAT_GAUGE_HEX=$(i2cget -y -f 0 0x34 0xb9)
BAT_GAUGE_DEC=$(($BAT_GAUGE_HEX))
```
Can you add these in?

sandeepmistry, 2017-02-16:
@norgeous pull requests welcome ...

norgeous, 2017-02-16:
I will attempt to get this done today (battery % only). I imagine that the code will be quite similar to `chipio.BatteryVoltage();` and will be a `five.sensor`. Will let you know how I get on.

norgeous, 2017-02-16:
Making a PR here https://github.com/sandeepmistry/node-chip-io/pull/43

norgeous, 2017-02-20:
#43 for battery percentage and now #45 for battery charge amps