sandeepmistry/node-chip-io #54: analogRead is not a function

ohpauleez, 2017-08-10:
First and foremost, thank you for all the effort you've put into this project.

When hooking up an FSR to the chip, I reached for J5's generic Sensor type (same general setup as found [in the docs](http://johnny-five.io/examples/sensor-fsr/)) - plugged into XIO-P0 and the setup like `let fsr = new five.Sensor({pin: "XIO-P0", freq: 25});`.

This results in error, "TypeError: this._chips[pinData.chip].analogRead is not a function" coming from [here](https://github.com/sandeepmistry/node-chip-io/blob/master/lib/chip-io.js#L139), triggered by J5's sensor support -- `johnny-five/lib/sensor.js:104`.

Any thoughts on why this might be happening?  Anything I can do to help out?  Need any additional information?

rwaldron, 2017-08-10:
[According to the readme](https://github.com/sandeepmistry/node-chip-io#pin-guide) there is only one pin that supports Analog input: `LRADC`

ohpauleez, 2017-08-10:
Gah, thanks @rwaldron!  Sorry about the noise.

closed 2017-08-10 by ohpauleez

rwaldron, 2017-08-10:
No problem ❤️