sandeepmistry/node-chip-io #55: Will this work on the Chip Pro board?

lastmjs, 2017-09-25:
Just wondering if Chip Pro is supported as well

lastmjs, 2017-09-25:
From some basic testing, it doesn't look like it works. These thread mentions some possible steps to take to get Chip Pro working with this repo: https://bbs.nextthing.co/t/installing-node-red-and-chip-io-on-chip-pro-new/17680/2

sandeepmistry, 2017-10-02:
@lastmjs I don't have a Chip Pro to test with, but this library can most likely support it if it's Allwinner R8 based, with some minimal changes and a Pro specific [pins.json](https://github.com/sandeepmistry/node-chip-io/blob/master/lib/pins.json) file.

Is there a way to detect at runtime if a CHIP for CHIP pro is being used?